rusefi/hellen-NB2-issues #5: Two rev B boards would not go into software DFU

rusefillc, 2021-07-09:
Two out of two rev D boards I have would not jump into SW DFU

HW DFU works fine
SW DFU just reboots the board into normal not DFU mode

@BeerMoneyMotorsports has the other three board in the mail so he would confirm on his end
hopefully @BeerMoneyMotorsports has a rev C to compare

BeerMoneyMotorsports, 2021-07-09:
@rusefillc i have rev C to compare

rusefillc, 2021-07-10:
i run experimenters with https://github.com/rusefi/rusefi/releases/tag/2021.06.28_release

i think _once_ it has rebooted into DFU but only once.

so it seems sporadic? maybe some power supply drama?

andreika-git, 2021-07-10:
Can you please make a close-up photo of the STM32 soldered on the board? We need to compare the chip labels with rev.C

mck1117, 2021-07-10:
I've just tried on a Rev A hellen 121 and it rebooted back to the firmware, not DFU

rusefillc, 2021-07-10:
@mck1117 welcome to the party!!! what about 121-vag-B?

@andreika-git any chance you can try Hellen-72 A or B you have?

rusefillc, 2021-07-10:
@andreika-git @mck1117 

H72-C is F429IIT6
H72-D is F407IGT6

mck1117, 2021-07-10:
I have just flashed `snapshot_20210628_214332_hellen121vag_rusefi` on an old Proteus 0.4 of August 2020 vintage, and "reset to DFU" button WORKED

rusefillc, 2021-07-10:
I wonder if there is a correlation or causation between exact CPU model and this issue

Which F4xx is Rev A hellen 121 vag?
Which F4xx is Rev B hellen 121 vag and how does it vote?
Which F4xx is old Proteus 0.4?

andreika-git, 2021-07-10:
> @andreika-git any chance you can try Hellen-72 A or B you have?
SW DFU works on H72b

rusefillc, 2021-07-10:
So one would suspect 407 right?

I've just tested with a small old DISCOVERY 407 and SW DFU works just fine? so 100 pin 407 is happy and 176 407 is not?

rusefillc, 2021-07-10:
@mck1117 could this be about power supply? especially my sporadic one time good reboot

maybe I shall add AP2337 tomorrow?

andreika-git, 2021-07-10:
Also you can try with an external +12V power

rusefillc, 2021-07-10:
@andreika-git is in the lead - rev D enters SW DFU as long as powered by +12v

mck1117, 2021-07-10:
It happens if connected to an actual USB host, but not if connected to USB power (usb power bank battery)

mck1117, 2021-07-10:
I've just tested with 12v connected, and it DID NOT HELP

rusefillc, 2021-07-10:
I've added one AP2337 to MCU module that made zero difference

I am not 100% but it looks like PROG holding + RESET does NOT always get me DFU only MOST of the time?! Need to made a video

mck1117, 2021-07-10:
I see the same thing, button is not 100% reliable either

rusefillc, 2021-07-11:
https://community.st.com/s/question/0D53W00000wyoGESAY/sporadic-reset-into-dfu-and-dfu-jump-any-links-between-power-and-dfu

rusefillc, 2021-07-11:
Matthew Kennedy  10:29 PM
I did some probing

The dfu software is running, then resetting after ~800us

Before the oscillator initializes

But it didn't even try the osc

[image]

rusefillc, 2021-07-11:
https://github.com/rusefi/hellen121vag/issues/50 was done while investigating this issue

mck1117, 2021-07-11:
Observation 1: powering from 12v (no USB connected), BOOT0 button is 100% reliable
Observation 2: powering from 5v USB power bank (no 12v, but also no data lines), BOOT0 button is 100% reliable.  Test performed with the same cable that is unreliable with laptop.
∴ the problem is not power, but is USB data lines.

mck1117, 2021-07-11:
H72B is 100% reliable with F427IGT6

mck1117, 2021-07-11:
Took the chip from the working H72B, put it on an H121B board.  Works perfectly.

mck1117, 2021-07-11:
DFU ROM matches, except for the odd duck that's a very old ST F4 discovery board:

[image]

rusefillc, 2021-07-11:
I've shared the drama with @dron0gus 

Dron says 407 has one VCAP while 42x have two VCAP or something? i like this line of inquiry

rusefillc, 2021-07-11:
With VAG and NA6 reports this is now https://github.com/andreika-git/hellen-one/issues/79

closed 2021-07-14 by rusefillc