rusefi/hellen-NB2-issues #8: OK let's try making Hall happier one more time

rusefillc, 2021-07-13:
With Frankenso we have somehow settles on 5K - 5K is mention even in the engine sniffer image file name on https://rusefi.com/forum/viewtopic.php?p=20509#p20509

Do we want to revisit one more time in light of https://github.com/rusefi/rusefi/issues/2958 logic analyzer images?

```
2:14
if you have a shitty falling edge, it's generally because the pull up is too strong (sensor can't pull it down), you have a bad connection on the ground pin (essentially a resistor in the ground path) or you have a shitty connection on the signal wire and your internal capacitances on your ecu circuitry can't be "emptied" by the effectively high resistance to the hall sensor

stefanst  2:16 PM
I tested Miata "Hall" sensors with 2k2 and 12V and they produce a nice, sharp edge.
```

@andreika-git what are current pull-ups? which PN can I try changing in your opinion?

andreika-git, 2021-07-14:
crank pull-up is 3.3k (R216)
cam pull-up is 4.7k

BeerMoneyMotorsports, 2022-01-07:
any update on this?  @andreika-git do you see this as actionable?  Do we need to make these changes on revE and other miata hellens?

BeerMoneyMotorsports, 2022-03-07:
closing due to no response and no functionality issues with NB2 rev D

closed 2022-03-07 by BeerMoneyMotorsports