rusefi/hellen-NB2-issues #13: H72d BOM shortage

BeerMoneyMotorsports, 2021-09-22:
Parts shortages as of 9/22/2021.  Support for alternatives needed!  plz halp

1- MWSA0503-100MT
2- NLV32T-2R2J-PF
3- LMR14020SDDAR
4- AUIPS2041LTR
5- FC-12M32.768KHZ7PF20PPM
6- STM32F427IGT6(Andrey supplies)
7- TJA1021T/20/CM,118

rusefillc, 2021-09-22:
@BeerMoneyMotorsports please add hellen designators to make things more efficient

how is NB1 doing? hellen designators are usually same between boards

BeerMoneyMotorsports, 2021-09-22:
NB1

1- MWSA0503S-100MT
2- LBC3225T2R2MR
3-  LMR14030SDDAR
4- ZXMS6005
5- SC-20S,32.768kHz,20PPM, 7pF
6- NA
7- not on BOM

rusefillc, 2021-09-22:
@BeerMoneyMotorsports we order parts at JLCPCB so using JLCPCB part numbers is pretty nice.

it looks like we have a solution for 80% of the ticket, let's close this one and open a new one if we have any remaining issues?

rusefillc, 2021-09-22:
See https://github.com/andreika-git/hellen-one/issues/72