rusefi/hellen-NB2-issues #23: Pin Error While driving

BeerMoneyMotorsports, 2021-11-16:
https://rusefi.com/forum/viewtopic.php?f=19&t=2190

shipping Matthew Kennedy the failed board via usps for inspection.

BeerMoneyMotorsports, 2021-11-20:
[image]

rusefillc, 2021-11-25:
Some relevant notes on https://github.com/rusefi/rusefi/issues/3558
There is an opinion that we need https://github.com/andreika-git/hellen-one/issues/118

mck1117, 2021-11-25:
I have the board in-hand now.

Notes:

- The STM32 gets hot when power is applied, but at least the core seems to still work, as TS connects and works (at least until it gets super hot and things stop working)
- Those pins that go to the TC4427 have 8-10 ohms to ground, and one of them has 2 ohms to the 3.3v rail, indicating catastrophic failures of the rail steering diodes (or possibly output transistors), and the other has ~20 ohms to the 3.3v rail.  

I'm not sure how that happened, but a 1k series resistor (optionally a pulldown too for safety) would likely prevent it by avoiding whatever overcurrent condition caused the steering diodes to fuse.

BeerMoneyMotorsports, 2021-11-25:
@mck1117 thank you.  Can you Write a ticket for us to add the resistor to future boards?  I need to order nb1 immediately. Is it too soon to make this call while you are still diagnosing everything?

Do we need to find a way to add this to current boards in customer hands?

mck1117, 2021-11-25:
is this not that ticket? https://github.com/andreika-git/hellen-one/issues/118

mck1117, 2021-11-25:
It will certainly not hurt.  Many proteus/mre/frankenso boards have been driving around with those resistors in place.

BeerMoneyMotorsports, 2021-11-25:
@mck1117 thankyou. Sorry I overlooked the ticket.

mck1117, 2021-11-25:
Actually I take that back - Proteus/MRE have pulldown resistors on the input side, but no series resistor.  They do, however, have a resistor on the output, also absent from Hellen.

BeerMoneyMotorsports, 2021-12-03:
Per @mck1117 reverse engineering OEM ecu.  He found a 1k resistor in series and a 100nf to gnd.  Picture below to show circuit.  This is being added to NB1 production.  NB2 is being tested with updated circuit.

[image]

related ticket-  https://github.com/rusefi/hellen-NB1-issues/issues/25

BeerMoneyMotorsports, 2021-12-03:
Modified board to test.

@mck1117 or @andreika-git , i would need some help with how to torture testing NB2.  

[image]

BeerMoneyMotorsports, 2021-12-21:
Board was tested for 250 miles then returned to the customer for further testing.  No failures during my testing which mainly consisted of lots of idling and driving.  This is with AC, lights, wipers, etc all running.

BeerMoneyMotorsports, 2022-01-07:
This board has 750 miles on it with zero problems.  Fix has been applied to rev nb2 rev E and then NB1 rev B.  Miata NA does not have alternator control circuit.  so 90-95 and 96/97 will have the 0.3 output module with protection but not a specific alternator control circuit.

closed 2022-03-07 by BeerMoneyMotorsports