rusefi/hellen-NB2-issues #25: 94 pin connector - position on board

andreika-git, 2021-12-15:
We need to determine the exact position of the new 94 pin connector.
The old 72 pin connector was not centered:
[image]
The new connector is obviously wider, and we need to know it's location to fit inside the case.

andreika-git, 2021-12-15:
@BeerMoneyMotorsports Can you please help?

BeerMoneyMotorsports, 2021-12-15:
We are using the exact same case as NB1 for this revision.  board footprint and connector positioning should match.  i can take precise measurements, however the nb1 perfectly fits the hole.

andreika-git, 2021-12-16:
The standard 64-pin connector is somewhat wider than the 94 pin connector:
[image]
So if we use the original NB1 case for the 94pin connector, then it would have ~3mm gaps on both sides. Is it ok?

Also, the NB1 case has different mounting holes (those on the sides). Should we use the NB1-style six holes or NB2-style five holes?

Can you please post a photo of the case which will be used for the 94-pin NB2 board?

BeerMoneyMotorsports, 2021-12-16:
Yes, that is correct on the connector size. My intention would be to have a thin printed piece to fill any excessive gaps. For the board, use NB1 footprint. My intention is to make it the same. This change will allow us to have 1 case for three different boards. Ill grab a pic today.

BeerMoneyMotorsports, 2022-01-07:
closing as we have moved from 94-122 pin.

closed 2022-01-07 by BeerMoneyMotorsports