rusefi/hellen-NB2-issues #26: 94 pin connector - row pitch conflict (footprint vs 3d-model)

andreika-git, 2021-12-16:
We have two options for the 94 pin connector: 1473181-1 and 1123794-2. 
According to the Both have the same row pitch as the 72 pin connector had:
[image]
The problem is that the 3D models given by the manufacturer don't match the footprint because they have different raw pitch (2.5 mm):
[image]

@BeerMoneyMotorsports Can you please measure the distance between the rows (of the pcb pins) on a real 94 pin connector? Is it 3.6mm+4.4mm or 2.5mm+2.5mm?

BeerMoneyMotorsports, 2021-12-17:
@rusefillc do you have a 96pin connector you can measure?  i have not purchased any yet.

rusefillc, 2021-12-18:
@BeerMoneyMotorsports @andreika-git I have a connector in my hands its 3.6mm+4.4mm

BeerMoneyMotorsports, 2022-01-07:
closing as we have moved from 94-122 pin.

closed 2022-01-07 by BeerMoneyMotorsports