rusefi/hellen-NB2-issues #27: NB2 rev.E schematics review

andreika-git, 2021-12-18:
Now we have a draft of the rev.E schematics.

How do we use an options connector (1A-1X, 22 pins)?
My proposal is to use it for:
- WBO: 6 pins
- ETB: 2 power pins and 8 analog pins (incl. +5VA and GNDA)
- An additional low-side output (boost?)
- An additional digital input (flex?)

[image]

Currently we use the existing rev.D pin mapping for the rev.E as is (so that our current Hellen72 config will work without changes), and adapt the pinout for ETB (we need three PWM pins for it) - there will be no connected 2E,2F,2H,2I pins (no EGR valve control).

@BeerMoneyMotorsports Can you please review the draft schematics?

andreika-git, 2021-12-23:
@BeerMoneyMotorsports Can you please comment on the current situation? It seems that we won't make it till christmas?

BeerMoneyMotorsports, 2021-12-23:
Here is what i would like to see in the options port.  All voltages need dedicated protection for user errors in wiring. 

Id like to leave the OUT_ETB +- in the molex connector.

1- WBO H-
2- WBO H+(12v out)
3- WBO R Trim
4- WBO Ip
5- WBO Vs/Ip
6- WBO Vs
7- Ground
8- 5v
9- 5v
10- 12v
11- 12v
12- Boost Controller
13- Digital Input(Flex fuel)
14- Pressure Sensor
15- Temperature Sensor
16-  Ground
17- IN_TPS
18- IN_TPS2
19- IN_PPS
20- IN_PPS2
21- 5v
22- Ground

Currently reviewing the total pinout.  We need to make sure some of the pins support normal NB2 and MSM.

BeerMoneyMotorsports, 2021-12-24:
**for msm** 4R and 4N should be tied together.  rather than using VVT on 4R, we need to use the second IAT as the primary IAT which is on 4R.  
[image]

BeerMoneyMotorsports, 2021-12-24:
the blue squares are 0R jumpers.

andreika-git, 2021-12-24:
> Here is what i would like to see in the options port. All voltages need dedicated protection for user errors in wiring.

Can you please edit your post and add "analog" or "digital" to all 5V and GND? They are different. (For ETB, we need 2 analog +5VA and two analog GNDA). Also we need at least one digital GND.

andreika-git, 2022-01-05:
Schematics & pcb were updated and ready for review!
What did i forget? :)
[hellen72_NB2.pdf](https://github.com/rusefi/hellen-NB2-issues/files/7816712/hellen72_NB2.pdf)

BeerMoneyMotorsports, 2022-01-07:
so far, almost everything looks ok.  I have some concerns for 4D that will require a jumper at minimum.  On MSM, it mentions a Wastegate Regulating Valve that i am unsure of.  Working now to identify it.  

On the Options Port, it isnt fully populated yet.  Missing some key things like dedicated Temp and Pressure?

BeerMoneyMotorsports, 2022-01-11:
MSM uses a factory boost control(wastegate regulating valve).  Pin 4d is the output so we need to have the boost control output T off to it with a jumper.  it will need to be removed for VVT.  We also need a jumper to disconnect the 12v on 4d when on an msm.  

tie 5k to 4d.  both need jumpers to disconnect depending on application.  

[image]

BeerMoneyMotorsports, 2022-01-11:
Other than the above, completion of the Options port seems to be all that is left.

BeerMoneyMotorsports, 2022-02-28:
@andreika-git 

Pin **4X- IN MAF- in_map1** is not needed.  please remove the connection to pin 4x.  This will be used by the options port. This will not effect H72d and older as MAF is not used.

I do not have a specific plan for some of these connections but i saw that they were available and would rather have them connected to the options port rather than just a blank pad.  The rest of the options port looks fine.  

5A- in_map1
5B- in_res3
5C- in_d4
5D- in_sens3
5E- out_solenoid_a1
5F- out_high1
5L- 5v

andreika-git, 2022-03-05:
Done! Can you please review the schematics and board files?
[image]

BeerMoneyMotorsports, 2022-03-05:
it looks good.  i cannot see any issues.  I see you already modified r23, so awesome!  Is it time to finalize?