rusefi/hellen-NB2-issues #28: Convert to 122 pin header

BeerMoneyMotorsports, 2021-12-24:
Lack of stock on 94 pin connector is forcing us to use the next size up, 122 pin connector.  It is a perfect fit for the camry case.

[image]

https://www.te.com/usa-en/product-8-5353015-1.html

andreika-git, 2021-12-24:
We need to know the exact position of the connector on board.

rusefillc, 2021-12-24:
nice find!

https://octopart.com/search?q=5353015&currency=USD&specs=0

[image]

BeerMoneyMotorsports, 2021-12-24:
Centered, connector in pic below is the same as 122 pin.

[image]

andreika-git, 2021-12-25:
Can you please measure the distance between the outer mounting points?
[image]

The is a misalignment between the official 3D model and the footprint (based on the datasheet):
[image]

BeerMoneyMotorsports, 2021-12-26:
@andreika-git i am measuring 111.1mm

BeerMoneyMotorsports, 2022-03-07:
closing.  completed with REV E

closed 2022-03-07 by BeerMoneyMotorsports