rusefi/hellen-NB2-issues #32: 3/7/22 rev E parts shortages

BeerMoneyMotorsports, 2022-03-07:
List of parts shortages.  Please assist with potential alternatives.
     Designator-- Mfr Part-- JLC part-- detail
1- D1,2,10,11-- C2297-- 0805 Light Emitting Diodes (LED) ROHS green
2- Q1.Q2-- AUIPS2041LTR-- C113610-- OT-223 Power Distribution Switches ROHS
3- F4, F5, F6-- 12H1500D-- C182446-- 100A SMD Fuse 5A 6.3 1206 Fuses ROHS
4- U601-- AD8628ARTZ-REEL7-- C29309-- 850μA 2.5MHz 2.7V ~ 5V, ±1.35V ~ 2.5V 1 V/us Zero-Drift 
5- c300-- CL10A106MA8NRNC-- C96446-- X5R 25V ±20% 10uF 0603 Multilayer Ceramic Capacitors 
6- U460-- SN74LVC4245APWR-- C7859-- TSSOP-24 Receivers & Transceivers ROHS
7- U490, U491-- 74HC2G17GW,125-- C94189-- 74HC 1 Buffers，Non-OppositeSide 2 SOT-363 Buffers / Drivers
8- Q601-- VNP28N04-E-- C263002-- TO-220 Power Distribution Switches ROHS
9- Q300, Q301, ETC-- PJM2319PSA-- C411720-- 40V 3.1A 1.25W 80mΩ@10V,3.1A 3V@250μA P Channel

andreika-git, 2022-03-08:
I've created the new BOM version with the following replacements. Please use the updated BOM file!

1 - SZYY0805G (C434432)
2 - no replacements in stock; Andrey presumably has some of ZXMS6005SGTA in the stash
3 - S1206-F-5.0A (C553924)
4 - GS8551-TR (C157715) *MAYBE* - it's not the best replacement, it's better to ask @mck1117 if it fits...
5 - CL10A106MA8NRNC (C2922482)
6 - 74LVC4245APW,118 (C6091)
7 - SN74LVC2G17DCKR (C50714)
8 - Q601 is not needed, because Q600 is back in stock (C2649436)!
9 - WST4045 (C719073)

mck1117, 2022-03-08:
number 4 looks OK to me, very very slightly worse, but not really in ways we care about!

closed 2022-10-03 by BeerMoneyMotorsports