rusefi/hellen-NB2-issues #34: Q1, Q2 footprint revision

BeerMoneyMotorsports, 2022-03-28:
Q1 and Q2 should have a footprint that allows both the AUIPS2041LTR and ZXMS6005SGTA chips.  Currently it does not, as the Tab on the chip is being used.  The AUIPS2041LTR and ZXMS6005SGTA do not have compatible footprints if the Tab is being used.  AUIPS2041LTR connects pin 2 to Tab and ZXMS6005SGTA connects pin 3 to Tab.  

[image]

rusefillc, 2022-03-28:
See also https://github.com/andreika-git/hellen-one/issues/147

andreika-git, 2022-03-28:
done (for the next rev.F)
[image]

BeerMoneyMotorsports, 2022-03-28:
Either rev F needs to be released now or this fix needs to be applied to Rev E.  AUIPS2041 is unavailable in my searches.  The only availability is ZXMS6005SGTA.  I have just enough AUIPS2041 for the 5 NB2e i have in hand but will not be able to produce any after that.

BeerMoneyMotorsports, 2022-05-12:
@andreika-git this is one that needs urgent attention.

BeerMoneyMotorsports, 2022-10-03:
fixed with F

closed 2022-10-03 by BeerMoneyMotorsports