rusefi/hellen-NB2-issues #42: Trigger sync issue NB2

BeerMoneyMotorsports, 2022-08-22:
Car- NB2
ECU- Hellen 72 rev D
New Firmware- 8/12/22 release
Old Firmware- 4/22/2022

Customer reports that he is hitting a "rev limit" at 6500 rpm and TS shows Trigger Sync issue.  This occurred after updating firmware and no tune changes. 

Here is the log: https://rusefi.com/online/view.php?log=989
Tune: https://rusefi.com/online/view.php?msq=1174

rusefillc, 2022-08-22:
Can he go back current or prev release?

BeerMoneyMotorsports, 2022-08-22:
He can go back to the 4/22 release if needed for testing.

rusefillc, 2022-08-23:
yes, please test again with https://github.com/rusefi/rusefi/releases/tag/20220812_release and https://github.com/rusefi/rusefi/releases/tag/20220702_release

BeerMoneyMotorsports, 2022-08-23:
Sebastian tried firmware 20220702 last night with same result.  Tonight he will try older.

BeerMoneyMotorsports, 2022-08-23:
Looks like this was a case of coincidence.  He double checked some things to make sure and found a failing cam sensor.  replaced and all is well with 8,12 firmware.

closed 2022-08-23 by BeerMoneyMotorsports