rusefi/hellen-NB2-issues #44: Settings Configuration Error

EkTyler, 2023-10-02:
I'm getting a Hardware settings configuration error once I get to the Gauge display. Not sure how to fix

Im using a HellenNB1 RusEFI 
on Firmware: rusEFI Master .2023.07.26.hellen72.3763268790

Engine info: 1.8l Nb1 

**To Reproduce**
Once I create a NEW project I locate my Ecu and find the firmware on Tunerstudio (TS). Then I select weather im using Lamda or AFR, I select AFR. after Im done selecting AFR I test my communication tab and test successful. It loaded project and is reading the controller settings. Once its done reading controller settings it then comes up with the error. 

Below is what message im getting:

"the Following error was reported by RusEFI:
HARDWARE VALIDATION FAILED led: CRITICAL status: unexpected startup pin state PH8 actual valve=0 logical value=1 mode=0

Unless directed above, you do not need to reload firmware to fix this problem.

To resolve this you need to correct this error in settings then powercycle your rusEFI"

My question is how do i go about fixing this?

Below are some Pictures containing my problem.

[image]

[image]

rusefillc, 2023-10-02:
Sounds like you have NB1 hardware but you are trying to use hellen72 which is NB2 firmware on it?

Please use rusefi_bundle_hellen-nb1 firmware on NB1 ECUs

EkTyler, 2023-10-02:
Okay, I will try to change firmware. I do so by downloading rusefi_bundle_hellen-nb1 firmware and extracting the zip file then going to the console folder and launching RusEFI Console correct?

rusefillc, 2023-10-02:
@EkTyler yes

Since you are changing firmware type you would have to "switch to DFU mode" and them "manual update via DFU" to go around the firmware type mismatch protection

EkTyler, 2023-10-02:
Okay, Im on Rusefi console and ive just changed from "auto update" to switch to DFU mode. 

Where is the Manual update via DFU? all im seeing currently is "UPDATE FIRMWARE" 

thanks

rusefillc, 2023-10-02:
Did it actually switch to DFU mode? Do you have happy "stm32 bootloader" in device manager? What are the options on the drop down once you have happy "stm32 bootloader"?

EkTyler, 2023-10-02:
DFU mode is when the blue light stops blinking correct? And no I don’t
think I’ve seen “stm32 bootloader” will check on this and report back!

rusefillc, 2023-10-02:
@EkTyler please try another option: hold "PROG" (or BOOT) button on the board and tap RESET button on the board, that should go into DFU mode

let me close this issue since we are falling into personal technical support territory which I am unable to sustain. Please use community channels to get help.

closed 2023-10-02 by rusefillc

EkTyler, 2023-10-02:
Still no luck @rusefillc Blue light still blinking on board.